Yonder-Deep/Nautilus #1: ***Checksum***

xuryann, 2021-03-07:
Used for error detection when communicating via radio

stephenbusi, 2021-05-30:
CRC 16 or 32

stephenbusi, 2021-05-30:
**Corrupted Messages**
BS -> AUV
Navigation Command:
- ignore the message
Mission Command:
- BS waits for response
- Send mission command again if Mission Status Data not in-progress

AUV -> BS:
- ignore
- Leave in format of    Position: X (Last received 85)

closed 2021-08-15 by stephenbusi

ServiceDog, 2021-08-15:
On closure, please reference the pull request number or commit hash which implements this feature request. Otherwise there's no indication that this feature was pushed to Main without investigating the code base

stephenbusi, 2021-08-16:
Continued in different issue ticket [here](https://github.com/Yonder-Deep/Nautilus/issues/4)