Yonder-Deep/Nautilus #64: Code refactoring

stephenbusi, 2022-01-24:
1. Separate Receive and Send thread classes into new files 
2. Create new Constants files
3. Remove unused functions, classes, and variables
4. Update code comments

stephenbusi, 2022-01-24:
Basestation Refactoring: https://docs.google.com/document/d/1O70Ru1EJTspxtO3G0YvSOusL-K6dkPmfdE5f6VheGdY/edit?usp=sharing
AUV Refactoring: https://docs.google.com/document/d/18gTte-em_Ju0GLhEE36lH2hEC6RCJPoBbuAqY3sYCA8/edit?usp=sharing

ServiceDog, 2022-01-24:
If multiple files declare gpio pin assignments, might want to split off a gpio config file so they're on the same page, easier to move pins around if needed

stephenbusi, 2022-01-28:
Looks like right now, we only reference gpio pins in the motor_controller file. If we add more references in other files later on, we'll make a gpio file then.

PR #65 (2022-01-31): 64/code refactoring
closes #64 
- add threads and static folders
- refactor run methods to call sub-methods
- remove unused variables
closed 2022-01-31 by stephenbusi